Tinkerforge/esp32-firmware #14: Investigate IP forwarding between network interfaces.

rtrbt, 2021-11-12:
This would allow connecting one charger to the WiFi AP of another, connecting the second charger to a WiFi network and then be able to reach both chargers over the network.

There seems to be some routing issues. A ping reaches another device in the charger's AP but the response gets lost.

5wA6as, 2024-06-29:
[https://github.com/espressif/arduino-esp32/issues/6421](https://github.com/espressif/arduino-esp32/issues/6421) is closed now.

Please reinvestigate the forwarding issue.

tillsteinbach, 2025-04-29:
Hey team, can this be reinvestigated? This would be totally a game changer for warp charger. My parents Tesla wants WiFi all the time for updates and I cannot reach the spot where the car is parked.
This would be a great USP in particular for people charging in an underground garage where the car cannot be reached due to the lack of cellular connectivity.