Tinkerforge/esp32-firmware #3: NFC card tracking collection of ideas

rtrbt, 2021-11-12:
- Use EVSE as RAM extension to keep track even if the ESP restarts
- Track charged energy per card
- Track charges per card; Timestamps? -> Requires NTP
- Allow setting max charging current per card
- Have a master card that can not start, but can stop charges started by any other card
- Generate reports? -> ESP does not have enough resources to generate PDFs but we could do this in web interface, i.e. in Javascript

jensstark, 2021-11-16:
Reports? Generate CSV or similar to create accounting records.
Running number, Tag ID, start time, end time, charged energy...

Retrievable via Web API, offering sample LibreCalc instructions to use.

Alestrix, 2021-11-23:
Idea: Once 1p3p is possible (Energy Manager), allow setting 1p/3p per nfc tag

borg42, 2025-07-13:
By now all of this is either implemented or possible with automation rules.

closed 2025-07-13 by borg42