Tinkerforge/esp32-firmware #67: Add OCPP support

rtrbt, 2021-11-16:
- OCPPJ 1.6
- Either implement it ourselves or use https://github.com/matth-x/ArduinoOcpp

kaeferfreund, 2021-12-26:
Would be very welcome 🙂

Wir haben im Moment 4x WARP Pro und würden uns sehr darüber freuen

kaeferfreund, 2021-12-26:
https://www.pv-magazine.de/2021/12/09/chargefriends-startklar-volle-watt-voraus/

ralf-freihaendig, 2022-11-04:
OCPP 1.6-J(SON) 
oder noch besser OCPP  2.0 wäre sehr wünschenswert. 
Viele aktuell verbauten Controller (Bender) nutzen noch das (bei Transactions) sehr umständliche OCPP 1.6 SOAP oder JSON.  

Wäre ein großer Schritt vorwärts! 
PS.: Danke für das tolle Produkt!

borg42, 2022-11-04:
@ralf-freihaendig Schau mal hier: https://www.tinkerunity.org/topic/8915-warp2-beta-firmware-mit-ocpp-unterst%C3%BCtzung/

rtrbt, 2023-03-23:
Fixed with 2.1.0 for WARP2. Difficult for WARP1. See here: https://www.tinkerforge.com/de/blog/new-functions-for-warp-charger-and-warp-energy-manager/

closed 2023-03-23 by rtrbt