Tinkerforge/esp32-firmware #76: Investigate HTTPS/WSS support with custom certificates

photron, 2021-11-23:
See https://www.tinkerunity.org/topic/7133-warp1-web-interface-via-https

generalmanager, 2021-12-13:
I'm really looking forward to this, ideally with HSTS or another form of certificate pinning.
Additionally it'd be great if the user could upload their own certificate, which can be derived from a known trusted root certificate. That's especially importand in corporate settings.

HPKP might also be nice to have:
https://serverfault.com/questions/876798/how-to-make-it-work-certificate-pinning-hpkp-and-self-signed-certificate-in-a

pottio, 2022-04-14:
Is there a timeline for when the features will be implemented?

rtrbt, 2022-04-19:
Not yet, sorry. As a workaround for now, you could use a reverse proxy. See for example here: https://www.tinkerunity.org/topic/7500-proxy/?do=findComment&comment=40444