Tinkerforge/esp32-firmware #84: some npm packages are deprecated

stif, 2021-12-10:
not sure if there are breaking changes if the packages are upgraded..
```
npm WARN deprecated urix@0.1.0: Please see https://github.com/lydell/urix#deprecated
npm WARN deprecated resolve-url@0.2.1: https://github.com/lydell/resolve-url#deprecated
npm WARN deprecated chokidar@2.1.8: Chokidar 2 will break on node v14+. Upgrade to chokidar 3 with 15x less dependencies.
npm WARN deprecated popper.js@1.16.1: You can find the new Popper v2 at @popperjs/core, this package is dedicated to the legacy v1
npm WARN deprecated core-js@2.6.12: core-js@<3 is no longer maintained and not recommended for usage due to the number of issues. Please, upgrade your dependencies to the actual version of core-js@3.
```

rtrbt, 2021-12-13:
All of those dependencies except popper.js are dependencies of various gulp modules, so they only relate to the build process, not the shipped web interface. We want to migrate away from gulp in the future anyway.

popper.js will be updated when we move to Bootstrap 5 (See #44).

closed 2021-12-13 by rtrbt

stif, 2021-12-13:
sounds reasonable, thanks for the update