Tinkerforge/esp32-firmware #85: Event Log: Convert timestamps to user-readable dates and times.

rtrbt, 2021-12-13:
There are two possible solutions for this:

1. Additionally to the uptime add a counter that counts overflows of the uptime, then use this counter to calculate a corrected timestamp (maybe even when inserting into the log, not in the frontend). The frontend can then convert this to a date and time.
2. Use NTP/RTC if #26 is implemented to insert timestamps into the log. However this could cause issues with time jumps and/or NTP unavailability on start-up.

bs-github, 2022-01-06:
I'd like to see a 3rd option in which the time is automagically taken and updated from the connecting client(s).
I think this would be lightweight and hopefully as simple as elegant.

The reasoning behind this would be:
- The local time of a connecting client is probably quite right (given that nowadays mobile phones as well as computers are internet connected and have their time from some NTP source).
- The WARPs do not have to be connected to the internet in case their time source is not necessarily on the internet.

One use case would be an underground parking space where reoccurring costs due to the need of an internet connection can be avoided using this third option.

rtrbt, 2022-01-07:
This is exactly what I've meant with option 1 :)
The issue there is that the uptime counter is a 32 bit variable in milliseconds, so it will overflow after ~50 days. So the ESP has to count the overflows to calculate a corrected timestamp (that can then be longer) when inserting into the log.

Option 2 does not have this issue, because we can just use the NTP time directly when inserting into the log. This also means user-readable timestamps everywhere the event log can be read, for example when calling /event_log directly or even on the serial console. Of course this requires a network connection, so probably we'll just implement both options.

bs-github, 2022-12-15:
I think this is already done, but I created a new issue that relates to this one.

[Add NTP server to RTC module](https://github.com/Tinkerforge/esp32-firmware/issues/187)

rtrbt, 2024-07-30:
Closing. The only thing missing from this issue is replacing timestamps relative to the ESPs boot with readable dates and times if neither NTP nor an RTC of some kind is available. I'd like to keep this behaviour as is, because this is the easiest way of noticing that the ESP does not know the (absolute) date and time.

closed 2024-07-30 by rtrbt