Tinkerforge/esp32-firmware #115: Idea: Allow meter/values_update topic

Alestrix, 2022-03-18:
Non-Pro WARP wallboxes have currently no way of interfacing with arbitrary external meters. If an external meter could "inject" its measurement values into the WARP using the proposed `meter/values_update` topic, then the Smart WARPs could offer the same charge tracking functionalities as the Pro versions.
The same could be done with `meter/phases_update`.

poohnet, 2022-03-21:
That‘s exactly what I‘m doing in my fork poohnet/esp32-firmware by implementing a new module called sdm630_mqtt…

poohnet, 2022-04-27:
Thanks @rtrbt, I've successfully tested this enhancement 👍 
The meter values are now available after changing my NodeRED flow accordingly (i. e. sending a "meter/state_update" followed by "meter/values_update", "meter/phases_update" and "meter/all_values_update"...

Alestrix, 2022-04-27:
Currently looking into (as for now only reading, no trying/testing) how to build this myself to play around with the new feature...

Shouldn't

> In software/platformio.ini backend (i.e. firmware) and frontend (i.e. web interface) modules can be selected to be compiled in the firmware.

in the project's main readme.md rather read "software/warp[2].ini"? As I don't see anything backend/frontend related in platformio.ini.

poohnet, 2022-04-27:
This has recently been changed. The different configurations have been split into these *.ini files which will automatically be included in main platformio.ini.

photron, 2022-04-27:
Nothing goes unnoticed here :smile:
Yes, @poohnet is correct, we just missed to update the instructions accordingly. I've fixed this now. Thanks for reporting @Alestrix.

commit 5656fc9b (2022-04-22): Add API meter module. Fixes #115.

This also stores unix timestamp of last reset: Fixes #27.
This was necessary to allow notifying an API meter that a reset was
requested.