Tinkerforge/esp32-firmware #131: Externe Steuerung funktioniert nicht zusammen mit Benutzerautorisierung - V2

bbr111, 2022-04-18:
Hallo,

ich habe die Externe Steuerung mit EVCC getestet. Externe Steuerung im WARP-Web-UI auf eingeschaltet.

evcc verion 0.90
Konfig auf warp-fw2 angepasst.

Die Wallbox ist eine WARP2 Firmware 2.0.1-624eeb29

Ist die Benutzerautorisierung aktiv, startet der Ladevorgang nicht.
Schaltet man die Benutzerautorisierung  ab, dann startet der Ladevorgang.

Was muss ich tun, damit beides aktiviert sein kann?

rtrbt, 2022-04-19:
Wenn du die Benutzerautorisierung und die externe Steuerung aktivierst, dann müssen beide den Ladevorgang freigeben, damit er gestartet wird. Du musst dann also sowohl eine NFC-Karte an die Wallbox halten, wenn ein Auto angesteckt ist, als auch die Freigabe von EVCC bekommen.

bbr111, 2022-04-19:
Das war in der alten Version anders.
Da konnte man beides zusammen verwenden. Die externe Freigabe hat ausgereicht um den Ladevorgang zu starten. Und zusätzlich war die Wallbox noch über NFC gesichert.

andig, 2022-04-19:
Wir könnten evcc dazu bringen, ein NFC Tag zu senden. Das wäre dann aber „statisch“, d.h. Evcc würde immer das gleiche Tag senden. Alternativ könnte Deine Steuerung das auch selbst machen.

Könntest Du den Usecase nochmal erläutern? Bedingt eine Freischaltung durch evcc nicht auch, dass ein passendes Fahrzeug angeschlossen ist? Falls ja könnte das dann auch gleich per NFC autorisiert werden.

bbr111, 2022-04-19:
Hallo Andig,

auch ne gute Idee, damit könnte man dann auf einen Benutzer alle Ladevorgänge von EVCC sammeln und auswerten.

Mein Anwendungsfall ist der, das ich eine mehr oder weniger öffentlich zugängliche Wallbox am Haus angebracht habe und nicht möchte, das sich Fremde dort einfach Strom zapfen können. Daher die Aktivierung der NFC Option.
EVCC und die Wallbox steure ich über HomeAssistant. Alles in einer Anwendung, das erhöht den WAF.
Ist unser Wagen angeschlossen, kann ich im HA einen Hacken setzten, das aktiviert dann EVCC und das Überschussladen per PV oder Min+PV. Ist der Wagen nicht angeschlossen, deaktiviere ich EVCC und Freunde und Bekannte könnten über NFC aktiviert laden. Demnächst eventl. noch der Firmenwagen dann per Abrechnung über eigene Karte.

Habe die Befürchtung das NFC beim Anstecken dann vergessen wird und man sich dann wundert, warum nicht geladen wurde.
Ich kann dann auch nicht mehr aus der Ferne nachsteuern, ohne die Benutzerautorisierung zu deaktivieren.

andig, 2022-04-19:
> Habe die Befürchtung das NFC beim Anstecken dann vergessen wird und man sich dann wundert, warum nicht geladen wurde.

Es bleibt das Problem, dass Du aus evcc nur 1 Tag injecten könntest.

> damit könnte man dann auf einen Benutzer alle Ladevorgänge von EVCC sammeln und auswerten.

Genau das ginge dann nicht mehr.

bbr111, 2022-04-19:
Aber nur für die EVCC Ladungen. Wenn ich EVCC über HA deaktiviere. ( Dienste per Script stoppen) Dann bleibt NFC ja weiterhin aktiv. Und die NFC User könnten gesondert ausgewiesen werden.

andig, 2022-04-19:
Wenn Du die Fähigkeit hast, dann kannst Du die NFC Tags aber auch gleich über HA hin schicken- dann brauchts keine Anpassung von evcc mehr?

bbr111, 2022-04-19:
ok, das muss ich mir mal anschauen? kann ich die NFC Freigabe per MQTT oder HTTP schicken? Dann würde das ja gehen.

bbr111, 2022-04-19:
nfc/inject_tag

Macht die Sache etwas komplizierter. könnte aber gehen.

Könnte man nicht auch in EVCC den Identifier des Vehicle als nfv/inject_tag senden?

andig, 2022-04-19:
> Könnte man nicht auch in EVCC den Identifier des Vehicle als nfv/inject_tag senden?

Das war ja mein Vorschlag. Aber welchen willst Du da schicken? Den des Autos? Das fände ich eine ziemlich charmante Idee, die Details müssten wir aber bei evcc diskutieren!

bbr111, 2022-04-19:
Machen wir. 
Denke den des Autos, man könnte ja mehre haben.

closed 2022-04-19 by bbr111