Tinkerforge/esp32-firmware #135: Investigate Home Assistant MQTT auto discovery.

rtrbt, 2022-04-27:
https://www.tinkerunity.org/topic/6431-warp-in-home-assistant-via-mqtt-und-http-api/#comment-41189

Seems to be supported in Home Assistant, openHAB and Domoticz. Implementing the auto discovery topics could be an easy way to add support for those systems. This would also make sure that the "bindings" match to the available API, as they are contained in the firmware itself.

iamthe1st, 2022-08-14:
I would love such an feature! To get readings of my Growatt Inverter, i use "Solar-Assistant.io" an Software for Raspberry Pi. 
This software supports MQTT discovery - with that the Home-Assistant setup was a matter of 2 Minutes, including about 20 Entities / Sensors.

rtrbt, 2022-11-22:
Fixed with 3a9906424bdf73105fca8689d2b140a71d12d3d3

bbr111, 2023-01-16:
@rtrbt  How to activate HomeAssistant Autodiscovery?

rtrbt, 2023-01-16:
Currently you have to compile the firmware yourself. Auto discovery is implemented, but not yet compiled into released firmwares, as we still have to finalize the MQTT topics.

mkaiser, 2023-03-22:
just installed it, enabled the MQTT Autodiscovery in the warp's webgui and.... tadaaaa, its working out of the box

BIG thank you!

[image]

captain186, 2023-08-07:
I can see the same topics as above, but I think there are more more available by the warp charger.
Aren't these submitted automatically? Can't find them in my Home Assistant entities.

If not, how do I get them into Home Assistant?
Thanks for any help!