Tinkerforge/esp32-firmware #170: Warp1: SDM630 not detected

22kW, 2022-10-30:
Hi, ich habe den SDM72 gegen einen SDM630 Modbus V2 im Warp1 charger (mit Firmware 2.0.7) getauscht, bekomme aber nur:
Found unknown meter type 0x0. Assuming this is a SDM72DM.
Leistung und Energie wird wir vorher mit SDM72 angezeigt, aber keine Ströme etc.
Habe nachgeschaut, SDM630 steht auf 3P4, woran liegt es ?

rtrbt, 2022-11-02:
Das ist ein Problem mit älteren Versionen des SDM630: Um rauszufinden, was für ein Stromzähler angeschlossen ist, lesen wir das Holding-Register 64515 aus, je nach Typ steht da 0x0084 (SDM72DM V1), 0x0089 (SDM72DM V2), 0x0070 (SDM630). Bei älteren SDM630-Versionen bekommt man aber nur eine 0.

Bei WARP2 behandeln wir das dann so, dass wir bei einer 0 auch den SDM630 vermuten. Bei WARP1 ist die Unterstützung von anderen Zählern als dem SDM72 V1 aber "nachgelegt", da hatten wir ursprünglich überhaupt keine Logik um den Zählertyp zu detektieren. Also darf auf keinen Fall die Unterstützung des SDM72 V1 gebrochen werden. Deshalb nehmen wir bei unbekannten Zählertypen _immer_ an dass es ein SDM72 V1 ist. Es könnte ja sein, dass es davon auch Versionen gibt, die nicht 0x0084 ausgeben.

Ich überlege mir mal eine Lösung für den alten SDM630. Entweder behandeln wir bei WARP1 auch 0 als SDM630 (das ist wie oben beschrieben aber etwas gefährlich), oder bauen einen manuellen Mechanismus ein, den Zählertyp zu überschreiben.

22kW, 2022-11-02:
das hatte ich befürchtet, dass der Zähler zu alt ist.
Wäre super wenn es da eine Lösung für early warp adopter geben würde.
Ich hatte halt den "alten" SDM630 noch, und wollte mich schon mal vorbereiten für eine 1/3 Phasen Umschaltung...

22kW, 2022-11-02:
Super, hat funktioniert.
Ich hatte nach Einbau vom SDM630 und vor dem type_override_update ziemlich viele der unten stehenden Meldungen.
Jetzt ist es weniger geworden, aber es gibt noch welche, was bedeuten diese ?

2022-11-02 18:19:52,315  Request 49: Exception code -1
2022-11-02 18:20:03,305  Request 227: Exception code -1
2022-11-02 18:51:29,049  Request 111: Exception code -1
2022-11-02 19:02:25,655  Request 251: Exception code -1

rtrbt, 2022-11-03:
Da das so selten passiert ist es vermutlich ein Timing oder Signal-Qualitäts-Problem. Stecken alle Drähte richtig und passen die RS485-Jumper? Du kannst auch mal versuchen zwischen Terminated und Non-Terminated zu wechseln. (siehe hier: https://www.tinkerforge.com/de/doc/Hardware/Bricklets/RS485.html#duplex-und-terminierung )

22kW, 2022-11-07:
Drähte und Jumper waren i.O. , evtl. Kabelbruch im starren RS485 Klingeldraht?
Bevor ich weiter suche: bei den "Request xx: Exception code -1" Einträgen geht es schon um RS485 Kommunikation ?

[debug-report-warp-SLC-2022-11-07T17-04-18-328.txt](https://github.com/Tinkerforge/esp32-firmware/files/9953392/debug-report-warp-SLC-2022-11-07T17-04-18-328.txt)

rtrbt, 2022-11-08:
Ja. Fehler der Form sind immer Timeouts (das ist die -1 in dem Fall) bei der Modbus-RTU-Kommunikation (also über RS485).

Es könnte auch sein, dass der Zähler einfach bei größeren Anfragen langsam antwortet und deshalb manchmal zu langsam ist. Teste bitte auch mal die angehangene Firmware (gezippt weil Github keine .bin-Dateien erlaubt, du musst sie vor dem Flashen entpacken), bei der ich die größeren Anfragen durch jeweils zwei kleinere ersetzt habe.

[warp_firmware_2_0_7_636a5e2b_merged.bin.zip](https://github.com/Tinkerforge/esp32-firmware/files/9961720/warp_firmware_2_0_7_636a5e2b_merged.bin.zip)

22kW, 2022-11-08:
scheint mir unverändert:
[debug-report-warp-SLC-2022-11-08T16-15-56-181.txt](https://github.com/Tinkerforge/esp32-firmware/files/9962541/debug-report-warp-SLC-2022-11-08T16-15-56-181.txt)

rtrbt, 2022-11-10:
Hm vielleicht doch ein Verkabelungsproblem? Alternativ muss ich mehr Logausgaben in die Firmware einbauen, damit wir ein besseres Bild vom Timing bekomen.

22kW, 2022-11-12:
Ich denke es liegt an meinem alten SDM630, ich glaube das war die erste Revision, die ist evtl. noch etwas buggy. Ich werde ausserdem das RS485 Kabel mal tauschen...

22kW, 2022-11-15:
Im Manual zum meinem alten SDM630 steht: das RS485 Kabel muss mindestes 1 Meter lang sein (!?)
Also habe ich ein neues geschirmtes längeres Kabel eingebaut: gleiches Problem.
Ich gebe auf. SDM630 vermutlich doof.

closed 2022-11-25 by 22kW