Tinkerforge/esp32-firmware #176: Modbus TCP: Decide whether we want to patch ESP-IDF to be able to get remote addresses.

rtrbt, 2022-11-18:
If we patch ESP-IDF to be able to access https://github.com/espressif/esp-idf/blob/6407ecb3f8d2cc07c4c230e7e64f2046af5c86f7/components/freemodbus/tcp_slave/port/port_tcp_slave.c#L74 from outside, we could show all active connections in the web interface. This would be helpful to debug Modbus TCP connection issues, as the Modbus master is typically a not easily debuggable appliance.

https://www.tinkerunity.org/topic/9215-warp2-beta-firmware-mit-modbus-tcp-wireguard-rtc-und-ocpp-unterst%C3%BCtzung/#comment-46549

kaeferfreund, 2022-12-10:
I would vote against it. Ist halt in der Zukunft ggf. mehr Arbeit wenn man vom Standard abweicht...

ffreddow, 2025-04-29:
We dont use the ESP-IDF Modbus/TCP implementation anymore

closed 2025-04-29 by ffreddow