Tinkerforge/esp32-firmware #181: Are the open topics WIP?

networxnet, 2022-12-02:
There is no new release for a longer time and no update at mikrocontroller.de thread. Therefore i wonder if there anybody is activly working on the open topics like RFID identification or MODBUS communication (EVCC connection?)?

borg42, 2022-12-02:
The last release was 4 days ago (https://www.warp-charger.com/warp2.html), we are very actively working on the firmware.

See here for the latest beta with new features: https://www.tinkerunity.org/topic/9215-warp2-beta-firmware-mit-modbus-tcp-wireguard-rtc-und-ocpp-unterst%C3%BCtzung/

rtrbt, 2022-12-05:
The mikrocontroller.net thread I've seen is about porting our firmware to other chargers. Maybe you are looking for this fork? https://github.com/bs-github/esp32-firmware

closed 2022-12-05 by rtrbt

bs-github, 2022-12-16:
There is a new pre release of my fork, but this is not the right place to talk about it.
https://github.com/bs-github/esp32-firmware/releases/tag/warpAC011K-2.0.8

bs-github, 2023-03-06:
Just if someone stumbles over this issue, the [warp-more-hardware](https://github.com/warp-more-hardware) project has released a new [version](https://github.com/warp-more-hardware/esp32-firmware/releases) of the ported software.