Tinkerforge/esp32-firmware #250: WEM reboots sporadically

deadrabbit87, 2023-06-14:
As described in the issue from [evcc](https://github.com/evcc-io/evcc/issues/8444), my WEM is sporadically rebooting. 

Therefore it is loosing the phase switching request from evcc. 

[debug-report-wem-26ud-2023-06-13T18-41-18-371.txt](https://github.com/Tinkerforge/esp32-firmware/files/11746567/debug-report-wem-26ud-2023-06-13T18-41-18-371.txt)

MattiasTF, 2023-06-14:
Ich hätte da ein paar Fragen zu deinem Setup:
Sehe ich das Richtig, dass du MQTT eingerichtet hast, um die WEM per evcc zu steuern und gleichzeitig Zählerwerte (nur power) in den WEM zu schieben, damit du eine bunte Grafik bekommst? Zusätzlich setzt du den Zählerstatus (meter/state_update) per HTTP-API? Falls ja, wie häufig tust du das und wie hast du das automatisiert?

deadrabbit87, 2023-06-14:
Hallo, 

Ja, das ist der Fall. Via Node-Red schreibe ich die Werte vom meinem Kostal Smart Energy Meter (KSEM) an die HTPP-API vom WEM: 
`.../meter/values`

Der Flow ist auch [hier ](https://www.tinkerunity.org/topic/11399-warp-energy-manager/?do=findComment&comment=50776) im Forum. 

Das passiert einmal pro Sekunde. Ich hatte das drin, weil die Steuerung mit EVCC zuerst noch nicht funktioniert hat. Wäre jetzt aber obsolet. 

MQTT ist wegen EVCC eingerichtet.

MattiasTF, 2023-06-15:
Ich habe leider kein Node-Red und versuche gerade, die JSON-Datei zu verstehen:

- Der obere Flow liest den Stromzähler aus und sendet den Leistungswert einmal pro Sekunde an `meter/values` und nicht `meter/values_update`?

- Der untere Flow fragt `meter/state` ab, und wenn das nicht `{"state":2,"type":4}` ist, wird es per `meter/state_update` auf genau das gesetzt. Wie häufig wird das ausgeführt? Auch einmal pro Sekunde?

deadrabbit87, 2023-06-15:
> * Der untere Flow fragt `meter/state` ab, und wenn das nicht `{"state":2,"type":4}` ist, wird es per `meter/state_update` auf genau das gesetzt. Wie häufig wird das ausgeführt? Auch einmal pro Sekunde?

Nein, einmal pro Minute. Der obere Flow ist richtig so, ja. Hier wäre auch noch ein Screenshot: 
[image]

poohnet, 2023-06-15:
Dieser sporadische Reboot tritt seit einiger Zeit auch bei meiner "WARP on Steroids" auf und auch ich sende die Zählerwerte periodisch per Node-RED.

So sieht der zugehörige Callstack des Crashs aus:
[image]

deadrabbit87, 2023-06-18:
Seit dem deaktivieren von Node Red sind meine Neustarts jetzt weg. WEM läuft bisher seit dem 15.06.2023 15:00 Uhr durch.

rtrbt, 2023-06-26:
Bitte nochmal mit Firmware 2.1.3 testen. Das Problem sollte jetzt weg sein.