Tinkerforge/esp32-firmware #288: Warp2 craches when using wss for ocpp on 2 operators.

filipvh, 2023-10-12:
I've tried two operators, one should be easy to test (ecarup.com).
In both cases, when using wss:// craches occured. The web interface was accessible for some seconds, so I could set ws://

rtrbt, 2023-10-12:
One of our chargers (running a nightly firmware) works fine with eCarUp and wss, so maybe this is a bug that has already been fixed? Please test again with this firmware:
[warp2_firmware-NIGHTLY_2_1_5_6527e1ff_72562fc53e6a071_merged.bin.zip](https://github.com/Tinkerforge/esp32-firmware/files/12882228/warp2_firmware-NIGHTLY_2_1_5_6527e1ff_72562fc53e6a071_merged.bin.zip)

(You have to extract the zip before uploading the firmware to the charger. Github doesn't allow me to upload the .bin file directly...)

filipvh, 2023-11-16:
I'm running the latest firmware 2.1.5-653faf69 now, and seems to be working fine now!

closed 2023-11-16 by filipvh