Tinkerforge/esp32-firmware #291: Get powermeter data via SML from powermeters (WARP Energy Manager)

janknieling, 2023-10-21:
Regarding this [blogpost](https://www.tinkerforge.com/de/blog/warp-energy-manager-released/) 

> SunSpec
Aktuell benötigt der WARP Energy Manager zum PV-Überschussladen oder für das dynamische Lastmanagement einen spezifischen Stromzähler am Netzanschluss. Wir wissen, dass oftmals bereits Zähler an diesen Punkten existieren, da Wechselrichter, Wärmepumpe o.ä. der anderen Hersteller ebenfalls hier Stromozähler fordern. Leider gibt es keinen generellen Standard den wir unterstützen könnten, der dafür sorgen würde, dass wir alle Stromzähler auslesen können. Zumindest gibt es mit SunSpec einen Standard, welcher von vielen Wechselricherhersteller unterstützt wird. Mittels dieser Schnittstelle kann dann der WARP Energy Manager die Daten kompatibler Stromzähler nutzen und es muss kein zusätzlicher Stromzähler verbaut werden.

the WARP Energy Manager can‘t collect the powermeter data from generic powermeters for dynamic load management or „PV Überschussladen“.
I think there is an easy possibility to collect data from generic powermeters via the SML protocol (https://de.wikipedia.org/wiki/Smart_Message_Language and https://www.bsi.bund.de/SharedDocs/Downloads/DE/BSI/Publikationen/TechnischeRichtlinien/TR03109/TR-03109-1_Anlage_Feinspezifikation_Drahtgebundene_LMN-Schnittstelle_Teilb.pdf)

The data from the powermeter can be collected via [volkszaehler](https://volkszaehler.org/) or [tasmota](https://tasmota.github.io/docs/Smart-Meter-Interface/).
I am using tasmota on an IR write-read head placed on my powermeter that reads out the data from my powermeters and publishing them to my MQTT broker.

It will be very nice if you can add support for collecting powermeter data from generic powermeters (with IR port and SML support) from an MQTT broker for dynamic load management and „PV Überschussladen“

andreas-bulling, 2023-11-25:
Hilft das? https://www.tinkerunity.org/topic/12038-startprobleme-mqtt-und-rs485modbus-mit-em540/#comment-53956

borg42, 2025-07-13:
Daten per HTTP/MQTT übertragen ist möglich, Tasmota ist auch möglich und hier dokumentiert: https://docs.warp-charger.com/docs/compatible_devices/devices#tasmota

closed 2025-07-13 by borg42