Tinkerforge/esp32-firmware #327: Add enable/disable function to wallbox users

batti, 2024-03-05:
Add a enable/disable function to be able to disable wallbox users temporarily.

janknieling, 2024-05-31:
I think it will be useful to drop all deleted users after implementing this (or provide a function to restore deleted user with 0A charging or disabled flag).
In my opinion it is better to provide a function to disable/enable users via flag instead of setting their allows current to 0A because it is clearer if you have a list of enabled and disabled users.
I have deleted a user max.mustermann some time ago because the user shouldn’t have the opportunity to charge. Now I would like to reautorize the user/recreate the user what is impossible because the charger said that there is a user with this name already existent

rtrbt, 2024-06-03:
> In my opinion it is better to provide a function to disable/enable users via flag instead of setting their allows current to 0A because it is clearer if you have a list of enabled and disabled users.

Good point. We probably should use a toggle switch in the UI (that still maps to 0 A on the API) and show disabled user accounts greyed out or similar.

> I have deleted a user max.mustermann some time ago because the user shouldn’t have the opportunity to charge. Now I would like to reautorize the user/recreate the user what is impossible because the charger said that there is a user with this name already existent

This is probably because there are charges tracked for this user. We store the user and display names of deleted users as long as there are still charges to be able to show them in charge logs. Currently there is no way to remove only some tracked charges (implementing this is part of https://github.com/Tinkerforge/esp32-firmware/issues/329 ), so to be able to recreate this user, you would have to remove all tracked charges.