Tinkerforge/esp32-firmware #329: charge_tracker: Migrate to 32 byte tracked charge entries. Then implement:

rtrbt, 2024-03-11:
- Flags byte/int
    - Entry is deleted
    - Start/Continuation (f.e. price change)
    - [...]
- Electricity price (See https://github.com/Tinkerforge/esp32-firmware/issues/328)
    -  If price changes during a charge, track a stop and a start
- Charge duration (accumulate time between B/C and C/B transitions)
- NFC-Tag-ID+Type?

rtrbt, 2024-08-27:
If this is done, add an automation action that sets the electricity price. This action can be used for day/night rates.

borg42, 2025-03-04:
We should also track "percent of pv-excess used" here.

lfrancke, 2026-09-01:
I'm very interested in this. I assume this is not news to you: Since the beginning of 2026 I need to prove which car charged in which session for tax reasons in Germany ([BMF-Schreiben](https://www.bundesfinanzministerium.de/Content/DE/Downloads/BMF_Schreiben/Steuerarten/Lohnsteuer/2025-11-11-selbst-getragenen-stromkosten.pdf?__blob=publicationFile&v=3)). The flat rates are gone :(

So I'd need the larger entries and the EV MAC in them.
I looked at this issue, at #328, #429 and the Eichrecht commits and [comment in the forum](https://www.tinkerunity.org/topic/13800-warp4-erster-eindruck-und-unterst%C3%BCtzung-f%C3%BCr-evcc/) to try and understand where you are.

Three things:

1. The vehicle should be a separate field from whoever authorised/started the charge (mentioning this here because this issue mentions the NFC Tag, which would not be enough to identify the car)

2. Both could reuse the identification types you already have in the EVSE bindings for "Eichrecht" (EICHRECHT_USER_ASSIGNMENT_IDENTIFICATION_TYPE_*: ISO14443, EMAID, EVCCID, ...). That would work for ISO 15118-2 as the EVCCID is the MAC address there. It'd break for -20 but the whole file format would break for -20 anyway as that might not even work with fixed field sizes so that's probably a totally different conversation.

3. On the migration I'd suggest just putting a format version in the filename. Nothing needs to be rewritten that way. And it could be done now and evolved again later. It wouldn't block a few use-cases that are useful on their own "only" for the Eichrecht case.

On the implementation side I'm afraid we're already well past 32-bytes with all the potential ideas. It's more like 48 (then file size wouldn't divide evenly by record size but that's okay, the code can be made to work with that) or 64 bytes (divides evenly but currently wastes a bunch of space). I don't actually know how large the data partition is. @poohnet in the forum mentions he has a fork and doubled the size of his records already and there's no space concern, so I'm hopeful. There are other options like variable length records etc. but a lot currently seems to rely on the fixed record length and it makes some things easier too.

So, I have a few concrete questions:
- How big is the data partition, and how much of it is free?
- Any update on your thinking on this issue you can share today? Are you still targeting 32 byte or are you considering a different format.
- What do you think about the "rough" idea above (version-in-filename plus 48/64 bytes and adding the MAC address)
- Is the signed Eichrecht dataset going to live inside these records, or alongside them? From the eichrecht_dataset/eichrecht_signature callbacks it looks like a separate blob but I'm just guessing.

I need this feature, I need it now ;-) so I'm happy to spend some time on this (I got a new Warp4 Pro) but I'd rather see/discuss what your current plans are first to see if there's any chance of getting something like this merged. No need for me to spend time on it if you have different ideas.

_Coincidentally, this is exactly what I do in my day job just at the exact other end of the scale with "big data", so a whole lot of this seems familiar to me.
And the research made me learn more about ISO15118-*, OMCF and all those protocols and formats than I ever wanted to know, thanks, that was a very interesting evening._ 😄

borg42, 2026-09-02:
@lfrancke Unfortunately, I think you’ve gone down the wrong rabbit hole here.

You can simply assign an NFC ID or MAC address to one user and call that user “EV A,” and assign another NFC ID or MAC address to another user and call it “EV B.” That gives you the differentiation between the two EVs that you’re looking for.

There is no legal requirement or anything similar that requires us to save the MAC address. In fact, the MAC address is completely useless for identifying an EV in the legal sense, since many EVs randomize it every time they plug in.

Independently of this, some employers in Germany require a charger to be Eichrechtskonform if you want to bill them for the kWh charged. Eichrechtskonformität is not a technical challenge. We already have everything required for it implemented. It is a legal and certification challenge.

To be able to sell an Eichrechtskonform WARP4, we need to be certified for Eichrecht Modul B (the charger itself) and Eichrecht Modul D (our production process). We are doing it and currently expect this to take around 6-9 months. As a small company, this will also cost us a significant portion of our yearly revenue just for the certification.

Once we are certified, we can sell an Eichrechtskonform version of the WARP4 Pro. However, if your charger is not public and your employer does not require Eichrechtskonformität, it offers no advantages over a regular WARP4 Pro.