Tinkerforge/esp32-firmware #338: Header fields are too long for server to interpret

alexander-zimmermann, 2024-04-28:
Hey all,

I move the WARP 3 box behind a Traefik reverse proxy to implement additional security like https (let's encrypt), MFA, WAF (Cloudflare), and so on.

If I'm accessing the Warp3 box now, I got the error message "Header fields are too long for server to interpret".

According to Google, we can fix that by bumping [CONFIG_HTTPD_MAX_REQ_HDR_LEN](https://github.com/espressif/arduino-esp32/blob/2.0.4/tools/sdk/esp32/sdkconfig#L801) to 1024.

Can you help me to test that? No clue were I need to put the CONFIG variable.

BR
Alex

rtrbt, 2024-04-29:
Hi Alex,

The maximum header length is already set to 1024: https://github.com/Tinkerforge/arduino-esp32/blob/d74a4d490aac9d67b6b041946a20ea568864ca29/tools/sdk/esp32/sdkconfig#L546

This is one of the ESP-IDF config variables. To change those, you would have to recompile the precompiled libraries that are embedded into the firmware. Doing that is a bit involved, but we have a (terse!) readme here, in case you want to test this for yourself: https://github.com/Tinkerforge/esp32-firmware/blob/master/software/lib-builder/README.md

Can you take a look at the HTTP requests that Traefik is sending? Which sent headers are that long? More than 1K seems to be a lot to me.

alexander-zimmermann, 2024-04-29:
Let me try to increase the debug lvl of traefik. If this doesn't help, I will try to run a tcpdump within the traffic docker container.

alexander-zimmermann, 2024-05-03:
The log output was not useful, but I played a bit around w/ treafik. I can already say that as soon as I enable Authelia in treafik, which leads to as session cookie, I run into the error. The following request header is already to much

```
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
Accept-Encoding: gzip, deflate, br
Accept-Language: de-DE,de;q=0.9
Authorization: Digest username="alexander", realm="esp32-lib", nonce="905b39b88f9b00000000000000000000", uri="/", response="a189a0a8feaa00000000000000000000", opaque="9e0e223633a900000000000000000000", cnonce="6e24a1efa65200000000000000000000", nc=00000001, qop="auth"
Connection: keep-alive
Cookie: authelia_session=CoyWq$YYYYY#F5wITkc#XXX#6e0s#XXX; cf_clearance=4Lx464zVSD2u91ol_8TfEZegT8mP.v0gVFXXRjRdfes-0000000000-0.0.0.0-hYYYYYYYYYYYYFmzNq.eCsax013RaT.VcPvj_gae2v1CO_mAnHjD44NqAm_zFyC7YWXXXXXXXXXXXXXXXXXXXX
Host: warp.zimmermann.sh
Sec-Fetch-Dest: document
Sec-Fetch-Mode: navigate
Sec-Fetch-Site: none
User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/17.4.1 Safari/605.1.15
```

alexander-zimmermann, 2024-05-03:
The log output was not useful, but I played a bit around w/ treafik. I can already say that as soon as I enable Authelia in treafik, which leads to as session cookie, I run into the error. The following request header is already to much

```
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
Accept-Encoding: gzip, deflate, br
Accept-Language: de-DE,de;q=0.9
Authorization: Digest username="alexander", realm="esp32-lib", nonce="[32 hex bytes]", uri="/", response="[32 hex bytes]", opaque="[32 hex bytes]", cnonce="[32 hex bytes]", nc=00000001, qop="auth"
Connection: keep-alive
Cookie: authelia_session=[32 chars]; cf_clearance=[150 chars]
Host: [19 chars]
Sec-Fetch-Dest: document
Sec-Fetch-Mode: navigate
Sec-Fetch-Site: none
User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/17.4.1 Safari/605.1.15
```

rtrbt, 2024-05-03:
Those are ~ 1000 bytes. So maybe it is enough if we bump the allowed header field length to 1536. You could also try to remove the unnecessary headers (for example the Cookie header, as we currently don't use any cookies) with Traefik: https://doc.traefik.io/traefik/middlewares/http/headers/#adding-and-removing-headers
(As far as I understand your setup, you only need those for the connection between your browser and Traefik, not for the connection between Traefik and the charger)

alexander-zimmermann, 2024-05-30:
Thanks for the support. Firmware 2.4.0 fixes the issue.

closed 2024-05-30 by alexander-zimmermann