Tinkerforge/esp32-firmware #341: evse/indicator_led API inconsistent between reading and writing

NobodysNightmare, 2024-05-03:
Hey there,
I am just in the process of setting up my Warp 3 and I noticed an inconsistency in the API to set the color of the LED. I've done everything through MQTT.

Reading the state topic `evse/indicator_led` I receive responses in the following format:

```
{"indication":1003,"duration":43,"color_h":120,"color_s":255,"color_v":255}
```

However, when trying to write that format into `evse/indicator_led_update`, I can see the following error message in the Warp logs (btw. it's awesome that I could find the error there!):

```
2024-05-03 12:10:42,726 | mqtt             | On evse/indicator_led_update: JSON object is missing key 'h'
JSON object is missing key 's'
JSON object is missing key 'v'
JSON object has unknown key 'color_h'.
JSON object has unknown key 'color_s'.
JSON object has unknown key 'colo
```

And sure enough, firing a different payload (an undocumented one) works fine:

```
{"indication":1003,"duration":30000,"h":120,"s":255,"v":0}
```

### Side-finds: Documentation

As an aside: I then had to trial-and-error my way through the allowed values for `s` and `v`. I wasn't sure whether their maximum is 100 or 255, but I guess it's 255 now. Also I am not sure what this means:

> Eine Helligkeit von 0 überlässt die Farbwahl dem Ladecontroller.

As far as I can tell, this resulted in my green (120°) to become yellow/orange-ish, so I am wondering whether that's working as intended.

I could create separate tickets for these issues, if that's helpful.

rtrbt, 2024-05-03:
Fixed, thanks!

>As far as I can tell, this resulted in my green (120°) to become yellow/orange-ish, so I am wondering whether that's working as intended.

This is correct, if you set 1003 as indication. 1003 is the "remind the user that a NFC tag is required" blink pattern which is orange by default. For example 1002 ("unknown tag") is red by default and 1001 ("known tag") is green.

If you want another color, you have to set the value (color_v) to at least 1. If you really want to disable the LED (any color with value 0 is black), set the indication to 0.

closed 2024-05-03 by rtrbt

NobodysNightmare, 2024-05-03:
Thanks for the explanation. Yeah, Hue was the only one that logicaly made sense. For S and V I was undecided between 100% and 255.

> If you want another color, you have to set the value (color_v) to at least 1. If you really want to disable the LED (any color with value 0 is black), set the indication to 0.

Oh wow, only now do I understand what's written there. The entire color choice is left to the controller, when the value is 0. I read this as "0 means the controller will chose the _value_ automatically", which makes a bit of sense when thinking about flashing lights...

Thanks a lot for clarifying this as well and thanks for the quick fix. Will need to remember to fix my home assistant integration after the next firmware update :)

commit f40d2428 (2024-05-03): evse_led: Fix color component naming. Fixes #341.