Tinkerforge/esp32-firmware #342: Safari: Password needs to be typed in again after each reload of the page

tillsteinbach, 2024-05-08:
On Safari (both Mac OS and iOS) I need to type in my password after every reload of a page of the web interface. On safari for iOS I also need to type in the credentials two times. After getting through the login I can browse through the web interface until I press the "reload" button again, or the page is reloaded by the browser.
At the same time using chrome on Mac OS is working very fine.
On my iPhone this bug is pretty annoying. Is there any information I can help with to fix this?

Best,
Till

rtrbt, 2024-05-13:
Sorry, investigating this took a bit. This is a (12 years old!) bug in WebKit (Safari's browser engine): https://bugs.webkit.org/show_bug.cgi?id=80362: WebKit does not use cached credentials if a WebSocket handshake gets a 401 response. For "normal" HTTP requests this works, but for the WebSocket handshake the browser immediately asks for credentials without checking the cache.

On macOS there is a work-around: If you press cancel when Safari asks for credentials the second time (or after reloading the page or restarting Safari). Safari will retry the request with the cached credentials. Unfortunately this does not work on iOS. I'm not sure if there is anything we can do to fix this. Maybe when we have implemented #76, we could switch to HTTP Basic Auth (we use Digest Auth at the moment to not send passwords via plain HTTP) but it looks like Safari has problems with that as well.

tillsteinbach, 2024-05-13:
Hmmm, bad news! Actually my usecase would be specifically iOS to control the Wallbox when standing outside. Maybe I have to try out another browser on iOS.

alexander-zimmermann, 2024-06-01:
@tillsteinbach You can move the esp behind a reverse proxy like Treafik (+ Authelia) and implement Basic Auth + MFA there

photron, 2025-05-06:
> * Safari shows the internal Basic Auth window on both macOS and iOS

This is intentional, as the custom dialog doesn't work properly with Safari in combination with WebSockets.

> * With Firefox on macOS/iOS, you see a proper login screen, but on iOS you're not redirected properly from it. At least it works on macOS.

I assume you see an empty web interface after the login screen? Empty in the sense, that the WARP header and the menu and content background are there, but the menu items and the actual content is missing. This happens when the WebSocket connection is missing. This is what happens in Safari when the custom dialog is used. It works on macOS because there Firefox uses it's own Gecko web engine, but on iOS Firefox is forced by Apple to use the Webkit web engine that is also used by Safari. This means on iOS Safari and Firefox show the same misbehavior. You should not get the custom dialog in Firefox on iOS. We detect Safari/Webkit based on the browsers user agent. What user agent does Firefox on iOS have for you? You can check with this website (the text in the blue box on the top):

https://www.whatismybrowser.com/detect/what-is-my-user-agent/

It should look somehing like this:

`Mozilla/5.0 (iPhone; CPU iPhone OS 14_6 like Mac OS X) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0.3 Mobile/15E148 Safari/604.1`

> * I also couldn't find a logout button

Logout cannot be triggered by the website with digest and basic auth, as the browser manages the login process.

stefanroeck, 2025-05-06:
My User Agent in Firefox on IOS looks like this:
`Mozilla/5.0 (iPhone; CPU iPhone OS 18_4_1 like Mac OS X) AppleWebKit/605.1.15 (KHTML, like Gecko) FxiOS/138.0 Mobile/15E148 Safari/605.1.15
`
In the mean time, I had to disable authentication completly because of issues with the http API.

Are there any plans to impelement a more sophisticated authentication handling with non-digest-based login/logoff logic? That would probably resolve most of these issues?

photron, 2025-05-07:
The detection didn't catch Firefox and Chrome on iOS correctly. I've changed that now. Here is a firmware to test. It has to be unzipped before flashing, because GitHub doesn't allow .bin files.

[warp3_firmware_2_8_0_681b2000_e46a7670f4c3e3b_merged.bin.zip](https://github.com/user-attachments/files/20080423/warp3_firmware_2_8_0_681b2000_e46a7670f4c3e3b_merged.bin.zip)

> Are there any plans to impelement a more sophisticated authentication handling with non-digest-based login/logoff logic?

There are plans to revisit this, but no road map right now.