Tinkerforge/esp32-firmware #345: Automatisierung lässt sich nicht speichern.

tillsteinbach, 2024-05-17:
Meine 3. Automatisierung lässt sich nicht speichern, auch die Fehlermeldung ist komisch weil sie "Lastmanager-Einstellungen" sagt.

[image]

Mögliche Gründe: Die Condition ist bei zwei Regeln die selbe?
Was total gut wäre, wäre wenn man mehrere Bedingungen verknüpfen könnte, oder auch mehrere Aktionen ausführen...

rtrbt, 2024-05-21:
Teste mal, ob du mit dieser Firmware
[warp3_firmware-NIGHTLY_2_3_0_664c598e_c8109c3e3dda834_merged.bin.zip](https://github.com/Tinkerforge/esp32-firmware/files/15387035/warp3_firmware-NIGHTLY_2_3_0_664c598e_c8109c3e3dda834_merged.bin.zip)
 speichern kannst (Erst entpacken, dann flashen. Github mag keine .bin-Dateien). Das Problem war, dass die WARP3-Änderungen bzgl. der LED (das ist jetzt eine RGB-LED statt einer einfarbig blauen) es nicht in die Automatisierungs-Aktion geschafft hatten.

> Mögliche Gründe: Die Condition ist bei zwei Regeln die selbe?

Das ist erlaubt und sollte funktionieren, damit kannst du dir also eine Regel bauen, die mehrere Aktionen ausführt.

Zwei Bedingungen verknüpfen geht im Moment nicht, unter anderem weil fast alle Bedingungen Ereignisse sind. Da müsste z.B. man erstmal definieren, was "gleichzeitig" bedeutet. Das hätte auch den Implementierungsaufwand extrem erhöht. Was wäre dein konkreter Use-Case dafür?

tillsteinbach, 2024-05-21:
Speichern ging jetzt, auch die Automatisierung "Fronttaster drücken -> Laemodus Schnell" funktioniert, das blinken der LED kommt nicht. Aber in der Firmware habe ich generell ein Problem mit den LEDs. Sie pulsiert eigentlich immer nur blau. Kein gelbes blinken wenn der Ladevorgang durch NFC blockiert ist und auch kein grünes blinken beim ranhalten des Tags.

Zum Use-Case:
Ich würde gerne mit dem Taster zwischen Schnell und PV wechseln, dafür bräuchte ich die Bedingung "Wenn Fronttaster Gedrückt && Lademodus Schnell" und "Wenn Fronttaster Gedrückt && Lademodus PV" für zwei Automatisierungen.

rtrbt, 2024-05-21:
Schalte unter Wallbox -> Einstellungen mal die externe Steuerung der LED ab. Die musst du nur aktivieren, wenn die LED über die API (nicht über z.B. die Automatisierungsregeln)  steuern möchtest.

> Zum Use-Case:
Ich würde gerne mit dem Taster zwischen Schnell und PV wechseln, dafür bräuchte ich die Bedingung "Wenn Fronttaster Gedrückt && Lademodus Schnell" und "Wenn Fronttaster Gedrückt && Lademodus PV" für zwei Automatisierungen.

Das haben wir prinzipiell vor, siehe #146. Wenn dir ein Wechsel reicht, also z.B. "Ich lade immer PV, es sei denn beim Anstecken des Autos (oder später) weiß ich, dass ich es heute eilig habe", dann könntest du zwei Regeln wie folgt schreiben:
- Fronttaster gedrückt -> Wechsle auf Schnell
- Ladestatus gewechselt von beliebig auf getrennt -> Wechsle auf PV

tillsteinbach, 2024-05-21:
Ahhh, das war mir nicht klar, dass die externe Steuerung die interne signalisierung komplett überschreibt. Danke dir!

Deine Regeln habe ich genau so drin (siehe Bild im Beitrag drüber), nur das zurückschalten geht so natürlich nicht, aber wenn man ehrlich ist, braucht man das auch nicht wirklich...

rtrbt, 2024-05-21:
> Ahhh, das war mir nicht klar, dass die externe Steuerung die interne signalisierung komplett überschreibt. Danke dir!

Das ist auch etwas "unterdokumentiert". Die Idee ist, dass wenn du eine externe Steuerung hast, die z.B. auch die NFC-Tag-Validierung übernimmt, wir dir da nicht dazwischenblinken wollen.

> Deine Regeln habe ich genau so drin (siehe Bild im Beitrag drüber)

Sorry, hatte ich übersehen.

closed 2025-04-29 by ffreddow