Tinkerforge/esp32-firmware #367: User Management Web Interface [enhancement request]

universal-dilettant, 2024-09-05:
Would be nice to have some rights control on the web interface. 

Example:

A **normal user** 
- can access the status page of the web interface and do the necessary adjustments while vehicle is getting charged: i.e. switch from "PV" to "schnell" etc. for the currently ongoing charging. 
- can set those values as default values (after being asked)
- can read access all other pages
- should be  password protected optionally only
- can be associated with an RFC token 

An **admin user** 
- can access the web interface as today
- should be password protected by default

Cheers

rtrbt, 2024-09-06:
That's on our list (even though I can't find another issue right now, so we keep yours). We already have a "roles" field per user, but currently don't do anything with it.

on-off-on, 2025-08-02:
any news within this implementation?