Tinkerforge/esp32-firmware #398: MQTT: Publish more information

othi, 2025-03-05:
Hi,

Would it be possible to publish more information via MQTT, specifically the allowed charging current and current charging mode?

Kind regards

rtrbt, 2025-03-05:
The allowed charging current is published as part of evse/state: https://docs.warp-charger.com/docs/mqtt_http/api_reference/evse#evse_state_any

The currently used charging mode is published as power_manager/charge_mode: https://docs.warp-charger.com/docs/mqtt_http/api_reference/power_manager/#power_manager_charge_mode_any

othi, 2025-03-05:
Hi,

You are correct. The reason I came to the conclusion that it wasn't is because Home Assistant did not show those values.

The autoconfiguration sent to Home Assistant (topic homeassistant/sensor/warp3-xx/chargerstate/config) is

```
unique_id: warp3-xx-chargerstate
object_id: warp3-xx-chargerstate
state_topic: warp3/xx/evse/state
value_template: '{{value_json.charger_state}}'
icon: mdi:ev-plug-type2
device:
  identifiers: warp3-xx
  manufacturer: Tinkerforge GmbH
  model: WARP3 Charger
  name: WARP3 Charger (warp3-xx)
name: Charge state
```

Would it then be possible to add those values to the Home Assistant autoconfiguration?

rtrbt, 2025-03-06:
> Would it then be possible to add those values to the Home Assistant autoconfiguration?
Sure