Tinkerforge/esp32-firmware #415: Phase Switching for BMW iX3 not working

marcelGoerentz, 2025-07-13:
Since the update to 2.8.5 the phase switching is not working like it was with 2.8.4. Seems like my BMW needs some extra time before connecting again with 3 phases.

marcelGoerentz, 2025-07-13:
@borg42 I will try to do so in the next few days.

borg42, 2025-07-13:
Thank you for the log! I i can see that the phase changing process was captured. Will take a detailed look tomorrow.

marcelGoerentz, 2025-07-21:
@borg42 Could you find anything in the logs? Or is there anything else I can help with?

borg42, 2025-07-22:
[warp3_firmware_2_8_6_687f61cf_8454797e8330767_merged.bin.zip](https://github.com/user-attachments/files/21364887/warp3_firmware_2_8_6_687f61cf_8454797e8330767_merged.bin.zip)

Can you test with this firmware? (you have to unzip it first, github does not allow to upload the .bin directly...)

marcelGoerentz, 2025-07-22:
Here the log, I can tell that it doesn't work for me.
The BMW didn't switch to 3p charging.

[warp3-2ePS-EVSE-Ladeprotokoll-2025-07-22T15-34-15-002.txt](https://github.com/user-attachments/files/21368257/warp3-2ePS-EVSE-Ladeprotokoll-2025-07-22T15-34-15-002.txt)

borg42, 2025-07-22:
But the same test does work with 2.8.4? Can you also make a protocol with 2.8.4?

This is your old charge protocol: https://vislog.warp-charger.com/WpTKZB9APxozDDYLz9jpLt
This is the new one: https://vislog.warp-charger.com/n5N3Nyj5bUC2FydkKrKkJy

If you disable everything but "Widerstand CP/PE (Ohm)" you can see what the BMW is communicating back to the EVSE.

With the old protocol you could see that after the CP disconnect was a small "glitch" by the EV with 880 ohm. The WARP Charger did react to it and close/open the contactor for a short period of time.
In the second test you can see that the cp disconnect lasts for 15s and there is no glitch. My idea was that the short closing/opening of the contactor was something the BMW didn't like. But it looks like this wasn't actually the problem.

The only difference to the 2.8.4 should be that the CP disconnect in the 2.8.4 was always 1 minute long by accident.

Is it possible that the BMW needs some ramp-up time to activate all three phases and you needed to wait a little longer to see that it charges with all three phases? We have seen with some EVs that always start with 1-phase, then go to 2-phase and then 3-phase.

marcelGoerentz, 2025-07-22:
I will do a test run with 2.8.4. I will ping as soon as I found the time for it.

I also think that this a timing issue. If I do the phase  switching manually, triggering the CP disconnect via API it works. But I didn't had the time to measure the sweet spot.

marcelGoerentz, 2025-07-24:
This is really weird now ...
Even with the firmware that worked, I can't successfully switch phases anymore...
But I've seen that on unsuccessful switching phases, p1 and p3 are active while p2 is still inactive (connected energy meter is telling me this).

marcelGoerentz, 2025-07-24:
I've been waiting for 30 minutes now. The car ist still charging with only 1 phase.  The second is active but there is no power flow.
Nevertheless, the car recognizes the 3 phases because its limits the max current to 16A. With single phase charging it is possible to consume 32A.

borg42, 2025-07-24:
> Nevertheless, the car recognizes the 3 phases because it limits the max current to 16A. With single phase charging it is possible to consume 32A.

Not sure what else we can do then. If you can find some timing/sequence to properly "reset" the BMW charge controller please let us know.

1 minute CP disconnect, which the beta firmware uses, is a long enough time that you could actually remove the Type-2 connector and plug in a different one. So i feel like this really ought to be long enough (the standard allows for a maximum reaction time of 3s...).

borg42, 2025-07-27:
> Maybe a stupid question. But I could it be an issue if the EV can't take 32 A with 3 phase, but the charger offers the 32 A?

I don't think so. The wallbox always gives the car the maximum amount of current is allowed to draw. The car decides how much it draws (up to the maximum).

> Another question is what does the sequence look like when faking the disconnection from the charger?

You can take a look at the exact details here: https://github.com/Tinkerforge/evse-v2-bricklet/blob/master/software/src/phase_control.c#L228

1. WARP Charger changes PWM to 100% and waits for the EV to respond by changing the resistance from 880 ohm to 2700 ohm. It waits for up to 6s here.
2. WARP Charger turns the contactor off
3. WARP Charger disconnects CP and waits 
4. WARP Charger changes the phases from 3 -> 1 or 1 -> 3
5. WARP Charger waits for 5 or 60 seconds (depending on if the EV did respond in 1.)
6. WARP Charger puts the PWM from before 1. on again.
7. WARP Charger waits for the EV to change resistance back to 880 ohm
8. WARP Charger turns contactor back on

In the test firmware that i posted the waiting time in 5. is increased to 15s and the other beta firmware that you tried had it accidentally always set to 60.

The EV is supposed to be able to react in 3s:

[image]