Tinkerforge/esp32-firmware #425: Implement support for 3rd party wallboxes load management via Modbus/TCP for WARP Energy Manager

janknieling, 2025-08-11:
It would be great to add the ability to manage third-party wallboxes with Modbus/TCP support via the WARP Energy Manager.
I have three wall boxes at home (two WARP Chargers and a KEBA P30 x Series from DKV for my company car) and I would like to control all of them via the WARP Energy Manager load management system.
My energy supplier states that the total consumption of all wall boxes combined must not exceed 22 kW.
My idea is therefore to collect the following data from the third-party wallbox via Modbus/TCP:

1. Charging state
2. Cable state
3. Charging current phase 1
4. Charging current phase 2
5. Charging current phase 3
6. Voltage phase 1
7. Voltage phase 2
8. Voltage phase 3
9. Phase switching state
10. Max charging current
11. Max supported current
12. Active power
13. Total energy
14. Serial number
15. Product type and features

(The above collection should be configured by the user, e.g. name, length, description, state [read-only or writable] and register via the configuration menu.)

The collected data should be processed together with the charging parameters/data from the WARP Charger for load management purposes.
The WARP Energy Manager will then set the required values on the third-party box via Modbus TCP (e.g. set the charging current and maximum energy transmission for the charging session, enable or disable the charging station and trigger the phase switch).

rtrbt, 2025-08-12:
One of my next "big" chunks of work is to decouple the charge management algorithm (that also handles PV excess charging) from the code that controls WARP chargers to allow controlling heating rods, heat pumps and other switchable loads by regarding them as some kind of charger. Supporting other chargers should then only require to implement the control interface, a Modbus register table in your case.

As the WARP charger can emulate the KEBA P30 c-Series register set, this will probably be one of the first third-party chargers to be supported.

janknieling, 2026-04-07:
Is there a timeline for the implementation of this?
It would be great to have the option of third-party wallbox load management via the WARP Energy Manager soon.